bigtreetech/Manta-M5P #1: Documentation not clear what are HV and VBB

grimlokason, 2023-02-12:
Hello,

I think that HV = high voltage and VBB = ?
But shouldn't it be stated that it's Voltage from  the HV-in pin or voltage from the card ?

( and for example, we can use a 24V for the mainboard, and a 2nd one dedicated to the driver ^^' )

maxkreja, 2023-02-16:
- `HV` equals any voltage you supply to `HV IN`
- `VBB` equals any  (as long as it's enough to power the board) voltage you supply to `DCIN`

You can choose which voltage is supplied to the stepper drivers by placing jumpers between two of the three pins between the driver socket and motor connector.

You have the following options:

- supply 24V to `DCIN` and nothing to `HV IN`:
    - `VBB` will be at 24V
    - `HV` will not have any voltage at all
- supply 12V to `DCIN` and nothing to `HV IN`:
    - `VBB` will be at 24V
    - `HV` will not have any voltage at all
- supply 24V to `DCIN` and 24V to `HV IN`:
    - `VBB` will be at 24V
    - `HV` will be at 24V too, but not drawing from `DCIN`
- supply 24V to `DCIN` and 48V to `HV IN`:
    - `VBB` will be at 24V
    - `HV` will be at 48V

This give you the option to use multiple power supplies to power the board and the motors. Or you can split the power between different rails on a single power supply (feed system from one rail, feed motors from other rail). And if you have steppers that require a different voltage than you want to supply to `DCIN`, you can do that too.

grimlokason, 2023-02-24:
Same here, and because i already had a octopus Pro.
But other users are not always familiar with it and i saw other request about it on Facebook.