bigtreetech/Manta-M5P #3: EBB36 CAN not working

zentaur1ger, 2023-02-15:
I've spent some time gathering information on how to get the Manta M5P to use the CANBUS with a CB1.
I first flashed klipper to the EBB36 according to the instructions in the manual.
Connected are 24V and CAN-H M5P<>CAN-H EBB36 and CAN-L M5P<>CAN-L EBB36.
LED on EBB36 is active.
The jumpers for the 120R resistor are set on both the M5P and the EBB36.
The CB1 also runs with klipper and I can also control all the pins from the M5P.
Then I have the entry in \boot\BoardEnv.txt

# uncomment the following overlays for MCP2515 spi to canbus module
overlays=mcp2515

activated and the CB1 restarted. 
The file nano /etc/network/interfaces.d/can0 was then created with the following content and restarted: 
auto can0 
iface can0 can static 
     bitrate 250000 
     up ifconfig $IFACE txqueuelen 1024 .
The selected bit rate also corresponds to that selected in the firmware for the EBB36.
However, the CANBUS does not start for me.
The output of dmesg looks like this:
biqu@TIDIRIUM-TT-SP:~$ dmesg | grep -i '\(can\|spi\)'
[ 1.257093] CAN device driver interface
**[ 2.267968] mcp251x spi0.0: MCP251x didn't enter in conf mode after reset
[ 2.268109] mcp251x spi0.0: Probe failed, err=110**
[ 2.268133] mcp251x: probe of spi0.0 failed with error -110
[ 2.277677] SPI driver fb_st7796 has no spi_device_id for sitronix,st7796
[ 2.329629] can: controller area network core
[2.329715] NET: Registered PF_CAN protocol family
[ 3.255536] mv64xxx_i2c 7081400.i2c: can't get pinctrl, bus recovery not supported
[ 3.304967] sun4i-drm display engine: [drm] Cannot find any crtc or sizes
[ 3.306270] sun4i-drm display engine: [drm] Cannot find any crtc or sizes
[ 364.607839] can: raw protocol

Elsewhere I found the information that
[ 2.267968] mcp251x spi0.0: MCP251x didn't enter in conf mode after reset
[ 2.268109] mcp251x spi0.0: Probe failed, err=110
indicates a connection problem and the error 110 represents a timeout. I checked the connection and it's fine.

And now, to be honest, I'm stuck. I have no idea if I need to configure anything else at any point.

Cisien, 2023-02-17:
@zentaur1ger I was able to get this working.

The detail that's missing from literally everywhere is that the Manta contains the U2C hardware built-in, and it's connected to the pi/cb over USB (internally)

You need to flash canboot/klipper to it as well.

The steps I performed:
1. `git clone https://github.com/Arksine/CanBoot.git` from /home/biqu
2. `cd ~/CanBoot`
3. `make menuconfig`
```
    Micro-controller Architecture (STMicroelectronics STM32)  --->
    Processor model (STM32G0B1)  --->
    Build CanBoot deployment application (8KiB bootloader)  --->
    Clock Reference (8 MHz crystal)  --->
    Communication interface (CAN bus (on PD0/PD1))  --->
    Application start offset (8KiB offset)  --->
(250000) CAN bus speed
()  GPIO pins to set on bootloader entry
[*] Support bootloader entry on rapid double click of reset button
[ ] Enable bootloader entry on button (or gpio) state
[ ] Enable Status LED
```
4. `make`
5. `cd ~/klipper`
6. `make menuconfig`
```
[*] Enable extra low-level configuration options
    Micro-controller Architecture (STMicroelectronics STM32)  --->
    Processor model (STM32G0B1)  --->
    Bootloader offset (8KiB bootloader)  --->
    Clock Reference (8 MHz crystal)  --->
    Communication interface (USB to CAN bus bridge (USB on PA11/PA12))  --->
    CAN bus interface (CAN bus (on PD0/PD1))  --->
    USB ids  --->
(250000) CAN bus speed
()  GPIO pins to set at micro-controller startup
```
7. `make`
8. Put the internal U2C into DFU mode: Press and hold boot, then press and release the reset button, release the boot button.
9. run `lsusb` to verify that there is an STM device in DFU mode (USB id should be `0483:df11`, if it's different, write it down and use it in the following two steps.
10. `sudo dfu-util -a 0 -D ~/CanBoot/out/canboot.bin --dfuse-address 0x08000000:force:mass-erase -d 0483:df11`
11. `sudo dfu-util -a 0 -d 0483:df11 --dfuse-address 0x08002000:force:leave -D ~/klipper/out/klipper.bin`
12. If not already done, configure `/etc/network/interfaces.d/can0` per the docs
13. Press the reset button (or power cycle the manta)
14. run `~/klippy-env/bin/python ~/klipper/scripts/canbus_query.py can0` and verify you have a canbus_uuid

When I flashed by EBB I flashed canboot using the dfu-util commands above. In this setup, the klipper image that will be flashed to the EBB needs to be configured with an 8k bootloader

zentaur1ger, 2023-02-18:
@Cisien 
First of all, thank you very much for your effort.
Yesterday I had already managed to get the CAN BUS to work. So I can now also address the M5P via CAN.
Note: According to everything I read yesterday, it is correct that if you flash the board with Klipper in such a way that you can use CAN, the serial connection is lost.
In the printer.cfg, the connection is now also defined via CAN.
The entry then looks like this:
[mcu]
canbus_uuid=17c3c36c0a13 (of course you have to use your own id)
canbus_interface: can0

The restart_method is considered invalid by Klipper in this setup and is therefore removed.
So now I can address the Manta itself completely.

However, now I face the problem that, no matter what I do, I don't see the EBB36 in the CANBUS.
The result is always
biqu@TIDIRIUM-TT-SP:~/klipper$ ~/klippy-env/bin/python ~/klipper/scripts/canbus_query.py can0
Total 0 uuids found

Only the IDs that are not configured in any way are displayed. As I understand it, this means that as long as the ID is not included in the printer.cfg and is addressed directly, it should now be displayed here.

Does the connection to the EBB work correctly for you and if so, how and with which settings did you flash it?

Cisien, 2023-02-18:
I came to the same conclusion with the can setup and also managed to get the ebb working. I was running into the same issue as you where i couldnt see the ebb no matter what i tried. I unfortunately fried the ebb when i was poking at it with a multimeter.

However, when i set up a new one it immediately worked after i flashed only klipper via dfu with the documented settings (no bootloader)

krompiel, 2023-03-28:
excuse me for butting in here. i'm looking to do a board swap also using m5p and ebb. from what i understand from this discussion the m5p does not need the separate u2c board?

zentaur1ger, 2023-03-28:
> excuse me for butting in here. i'm looking to do a board swap also using m5p and ebb. from what i understand from this discussion the m5p does not need the separate u2c board?

No, it is integrated in the M5P.

krompiel, 2023-03-28:
> > excuse me for butting in here. i'm looking to do a board swap also using m5p and ebb. from what i understand from this discussion the m5p does not need the separate u2c board?
> 
> No, it is integrated in the M5P.

YOU sir are a gentleman and a scholar.

Cisien, 2023-03-28:
The biggest hurdle for me to understand was that I needed to access the m5p by its canbus_id instead of serial after flashing the board with a u2c klipper

itz-magic, 2023-04-03:
Hey guys I have ran into an issue setting up my M5p board and EBB42 Canbus board. I know the EBB42 is Very similar to the EBB36 so I figured you guys could help. When querying for the UUIDs I am able to get the UUID for the klipper chip but I am unable to get the UUID for the Canboot chip. please help I do not want to have to buy a U2C to get this to work. Also where are you plugging in the EBB42 when you are flashing canboot? is it to the external Double USB A port or to the single USB A port next to the Canboot plugs ?

Cisien, 2023-04-05:
connect the EBB42 over USB and flash canboot with my instructions above. Disconnect from USB and connect back to CAN+power over the U2C

`make menuconfig`, select whatever applies to an EBB42
 - use the canbus interface and pins in the documentation.
 - select 8kb boot loader
`make`

Confirm you can see the canboot device with the canbus query command, get the id from there and use it to flash klipper using the following command:
~/CanBoot/scripts/flash_can.py -i can0 -f ~/klipper/out/klipper.bin -u <id from query>

If the flash stalls and errors out part way through, you need to update your u2c

put the u2c into boot mode by holding down the boot button as you plug in USB
```
wget https://github.com/Arksine/CanBoot/files/10410265/G0B1_U2C_V2.zip
unzip G0B1_U2C_V2.zip
lsusb and find the id of the u2c (probably the same as any other STM device) (0483:df11)
sudo dfu-util -a 0 -D G0B1_U2C_V2.bin --dfuse-address 0x08000000:force:mass-erase:leave -d 0483:df11
```
unplug the u2c, plug it back in, try the flash again
if that completes, your canbus query command should return the id of your ebb

itz-magic, 2023-04-06:
> connect the EBB42 over USB and flash canboot with my instructions above. Disconnect from USB and connect back to CAN+power over the U2C
>
> `make menuconfig`, select whatever applies to an EBB42
>
> * use [...]

what if Im trying  to connect to the onboard U2C on the M5P board? how would I go about disconnecting and connecting to that? 
Or did I miss read and I need to purchase a btt U2C and put candle light onto it?

Cisien, 2023-04-06:
Sorry for the confusion, I was using a U2C for my last project, so that was top of mind, you can disregard it if you're using the onboard u2c on the manta.

itz-magic, 2023-04-06:
> Sorry for the confusion, I was using a U2C for my last project, so that was top of mind, you can disregard it if you're using the onboard u2c on the manta.

could you run me through the process in detail. I have used your guide and still cannot get it to connect. I get the klipper uuid just fine but cannot get the canboot uuid to save my life. i wouldnt be pushing so hard if there was adequate documentation on it but your my best try.

itz-magic, 2023-04-06:
i can do that.

IniterWorker, 2023-06-01:
There is no mcp251x in the hardware sch file. The issue should be closed and renamed.
The user manual must be revised with the correct way to enable the USB to CAN.

zentaur1ger, 2023-06-01:
> There is no **mcp251x** in the hardware sch file. The issue should be closed and renamed. The user manual must be revised with the correct way to enable the USB to CAN.

What exactly does that mean then?

IniterWorker, 2023-06-01:
> > There is no **mcp251x** in the hardware sch file. The issue should be closed and renamed. The user manual must be revised with the correct way to enable the USB to CAN.
> 
> What exactly does that mean then?

See CM4 block in the [SCH file](https://github.com/bigtreetech/Manta-M5P/blob/master/Hardware/BIGTREETECH%20MANTA%20M5P%20V1.0-SCH.pdf). You will see that there is no CAN bus IC provisioned on it.

zentaur1ger, 2023-06-01:
@IniterWorker wait, so you're telling us, that there is a can bus connector on the M5P, but there is no chip available, that would be able to deliver the can signal?

IniterWorker, 2023-06-01:
@zentaur1ger, the easy way to see it:
- Only the MCU has the IC to use the CAN Bus connector.
- The CM4 (SOM/Linux) doesn't have any CAN Bus IC, the overlay `MCP251x` is not available, so the driver report a timeout `-110`.

IniterWorker, 2023-06-01:
@zentaur1ger,  the only way to use CAN bus on the MP5 is to enable https://github.com/bigtreetech/Manta-M5P/issues/3#issuecomment-1435406181 USB to CAN bus bridge. The reason why, it's because you don't have any component (IC) available on the CM4 hardware part. 

**Note:** If you enabled the overlay from `BoardEnv.cfg` you will have issue with the @Cisien configuration.

zentaur1ger, 2023-06-01:
@IniterWorker thank you for that clarification

McMabb, 2023-09-19:
Gentlemen, I did everything according to the instructions and I received this help message

pi@hypercube:~ $ ~/klippy-env/bin/python ~/klipper/scripts/canbus_query.py can0
Traceback (most recent call last):
  File "/home/pi/klipper/scripts/canbus_query.py", line 64, in <module>
    main()
  File "/home/pi/klipper/scripts/canbus_query.py", line 61, in main
    query_unassigned(canbus_iface)
  File "/home/pi/klipper/scripts/canbus_query.py", line 20, in query_unassigned
    bus = can.interface.Bus(channel=canbus_iface, can_filters=filters,
  File "/home/pi/klippy-env/lib/python3.9/site-packages/can/interface.py", line 127, in __new__
    return cls(channel, *args, **kwargs)
  File "/home/pi/klippy-env/lib/python3.9/site-packages/can/interfaces/socketcan/socketcan.py", line 594, in __init__
    bind_socket(self.socket, channel)
  File "/home/pi/klippy-env/lib/python3.9/site-packages/can/interfaces/socketcan/socketcan.py", line 459, in bind_socket
    sock.bind((channel,))
OSError: [Errno 19] No such device

IniterWorker, 2023-10-02:
> Gentlemen, I did everything according to the instructions and I received this help message
> 
> pi@hypercube:~ $ ~/klippy-env/bin/python ~/klipper/scripts/canbus_query.py can0 Traceback (most recent call last): File "/home/pi/klipper/scripts/canbus_query.py", line 64, in main() File "/home/pi/klipper/scripts/canbus_query.py", line 61, in main query_unassigned(canbus_iface) File "/home/pi/klipper/scripts/canbus_query.py", line 20, in query_unassigned bus = can.interface.Bus(channel=canbus_iface, can_filters=filters, File "/home/pi/klippy-env/lib/python3.9/site-packages/can/interface.py", line 127, in **new** return cls(channel, *args, **kwargs) File "/home/pi/klippy-env/lib/python3.9/site-packages/can/interfaces/socketcan/socketcan.py", line 594, in **init** bind_socket(self.socket, channel) File "/home/pi/klippy-env/lib/python3.9/site-packages/can/interfaces/socketcan/socketcan.py", line 459, in bind_socket sock.bind((channel,)) OSError: [Errno 19] No such device

Did you @McMabb follow the https://github.com/bigtreetech/Manta-M5P/issues/3#issuecomment-1572284173 ?

tripslip, 2023-10-06:
**I am struggling as well to get a uuid for the ebb.**
If Im understanding this correctly, I will only have 1 uuid and followed by Can0? If that's the case, what would my cfg look like to access ebb36 pins? I planned my whole build around using just one board and would love to see it work.

McMabb, 2023-10-09:
As you write, with Manta M4P + U2C + ebb36/ebb42.

From what I managed to determine, the Manta M5P should be switched to CAN0 mode and both drains must operate in this mode.

tripslip, 2023-10-13:
Finally got it working. I think it was the CB1 and the core 3566 that were the problem. I broke down and paid scalper prices for a cm4 and had it up in no time. Hard lesson learned.

Yumi216, 2024-08-12:
There are some problems with the steps here, I refer to the steps linked below, but the configuration is based on yours, and it was successfully installed.
https://forum.vorondesign.com/threads/manta-m8p-btt-ebb-sb22xx-setup.1697/