bigtreetech/Manta-M5P #7: ADXL345

MasterDoge1, 2023-03-03:
there is no wiring diagram for the adxl345 or pinout for this board available

jayden-jw, 2023-04-03:
Yes, it's not in the user manual. I wanted to know how to connect too.

Valkar13, 2023-05-14:
in example printer.cfg is
#[adxl345]
#cs_pin: PC0
#spi_bus: spi2]

but Klipper throw error  spi_bus: spi2] does not exist
What to do?

Vojtaaaaa, 2023-11-08:
Is there some solution?

AndylgTom, 2024-03-09:
How to make it work, any instructions ?
It's like this now and error:
status Klipper: ERROR
mcu 'mcu': Unable to connect

```
[adxl345]
cs_pin: PC0
#spi_bus: spi2
spi_speed: 5000000
spi_software_sclk_pin: PB13
spi_software_miso_pin: PB14
spi_software_mosi_pin: PB15
rate: 3200
axes_map: x,z,y
```

Valkar13, 2024-05-03:
1) disconnect ADXL345
2) boot up board with Klipper, (Klipper must work)
3) connect ADXL345
4) run tests