bigtreetech/Manta-M5P #11: MKS PWC 3.0 wiring and config

oleksii-suprun, 2023-06-12:
Hi All!

Did anyone connect MKS PWS 3.0 board to Manta M5P? I found out how to connect and configure PW-OFF in this topic (https://github.com/makerbase-mks/Klipper-for-MKS-Boards/issues/18#issuecomment-1110287092) and now I can power on my board using a power button, but I'm not able to power it off with that button. I don't understand how to connect PW-DET.

Thanks in advance. Any help will be much appreciated.

oleksii-suprun, 2023-06-23:
After some time, I identified, that my wiring was not 100% correct and did not work as expected. After some tries and failures, I identified the following problems:
1. CB1 module which I used together with my Manta M5P has a different pinout than RPi(CM4). 
2. Manta's GPIO has 3.3V logic, but MKS PWS is designed to work with 5V.

It was very challenging to find a free pin to connect the PWS module. The main problem was that not only naming of pins is different between RPi (CM4), but some of them are incompatible. The following links provide a bit of information on how to use GPIO with CB1 and which pins were already discovered:
- https://github.com/bigtreetech/CB1/issues/14
- https://github.com/bigtreetech/CB1/issues/44#issuecomment-1264738711
- https://github.com/bigtreetech/CB1/issues/44#issuecomment-1264741560

After some back and forth, I finally identified which GPIO pin I can use for wiring. But then it was required to resolve the issue with incompatible logic voltages (3.3V logic and 5V). I used the logic level converter module. Now everything works fine.

My `moonraker.conf`:
```
[power printer]
type: gpio
pin: gpio73
off_when_shutdown: True
initial_state: on
bound_services: klipper
```
 
My `printer.conf`:
```
[gcode_macro M81]
gcode:
  {action_call_remote_method(
    "set_device_power", device="printer", state="off"
  )}
```

Pamilias, 2024-06-19:
I tried this and measured 3.3v when on and 5v when off. It should be 0v when off right?