bigtreetech/Manta-M5P #15: USB Ports not working with CM4

Pieterv24, 2023-07-11:
I'm having trouble getting the USB ports to work on my M5P when using the CM4.
As stated in the manual I've added the following line to /boot/config.txt
`dtoverlay=dwc2,dr_mode=host`

when running `lsusb`i get the following output:
```
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```
This does not show the webcam nor the M5P itself.
I've also tried to use other usb devices which don't get detected.

output of `dmesg | egrep -i "dwc2|usb"`
```
[    0.144738] usbcore: registered new interface driver usbfs
[    0.144811] usbcore: registered new interface driver hub
[    0.144892] usbcore: registered new device driver usb
[    0.145300] usb_phy_generic phy: supply vcc not found, using dummy regulator
[    0.145552] usb_phy_generic phy: dummy supplies not allowed for exclusive requests
[    1.773244] usbcore: registered new interface driver r8152
[    1.773332] usbcore: registered new interface driver lan78xx
[    1.773420] usbcore: registered new interface driver smsc95xx
[    1.775000] usbcore: registered new interface driver uas
[    1.775093] usbcore: registered new interface driver usb-storage
[    1.786533] usbcore: registered new interface driver usbhid
[    1.786559] usbhid: USB HID core driver
[    5.047093] dwc2 fe980000.usb: supply vusb_d not found, using dummy regulator
[    5.056226] dwc2 fe980000.usb: supply vusb_a not found, using dummy regulator
[    5.114668] dwc2 fe980000.usb: DWC OTG Controller
[    5.114732] dwc2 fe980000.usb: new USB bus registered, assigned bus number 1
[    5.114799] dwc2 fe980000.usb: irq 36, io mem 0xfe980000
[    5.115361] usb usb1: New USB device found, idVendor=1d6b, idProduct=0002, bcdDevice= 6.01
[    5.115384] usb usb1: New USB device strings: Mfr=3, Product=2, SerialNumber=1
[    5.115400] usb usb1: Product: DWC OTG Controller
[    5.115413] usb usb1: Manufacturer: Linux 6.1.21-v8+ dwc2_hsotg
[    5.115426] usb usb1: SerialNumber: fe980000.usb
[    5.128312] hub 1-0:1.0: USB hub found
```
`ls -R /sys/ | grep dwc2`
```
dwc2
/sys/bus/platform/drivers/dwc2:
dwc2-unaligned-dma
dwc2
/sys/module/dwc2:
/sys/module/dwc2/drivers:
platform:dwc2
/sys/module/dwc2/holders:
/sys/module/dwc2/notes:
/sys/module/dwc2/sections:
dwc2
```

Any suggestions of what I can do to resolve this?
I'm having no trouble when a CB1 is plugged in.

DavidvtWout, 2024-01-12:
I got the exact same problem with a CM4 on a M8P. Only the root hub device shows up (compared to nothing at all when `dwc2` was not enabled). 

@Pieterv24 Did you find a solution?

alexvnesta, 2024-10-29:
same issue here as well. I am thinking it could be a hardware issue?

alexvnesta, 2024-10-29:
after re-seating my cm4 a few times, usb started working partially - in my case, this appears to be a hardware issue.