bigtreetech/Manta-M5P #17: Extruder Fan Not Spinning

Tommy1856, 2023-08-22:
I'll fill in the details when I get home but my Extruder Fans aren't spinning. I've tried tweaking the printer config with no luck. It appears to be getting power (Tested with a multi-meter). I've also replaced all other cables except the one that plugs into the board itself.

TheNitek, 2023-08-22:
Wrong polarity (a.k.a. pins mixed up)?

Tommy1856, 2023-08-22:
> Wrong polarity (a.k.a. pins mixed up)?

Yeah, I tried swapping them over but it shorted the board. On the original board it also broke randomly. I replaced it but the new one also doesn't spin. I plugged it in directly to the main board and it works fine, just not when plugged into its proper position.