bigtreetech/Manta-M5P #19: Struggling with M5P and CB1 eMMC

Limey343, 2023-10-26:
Good afternoon, I've got a brand new M5P with cb1 eMMC.  Firstly I was able to get the firmware to flash onto the M5P and after rebooting the power light would flash on and off constantly.  CB1 I had to send back and get a replacement due to missing antenna and it would not do anything at all. Got the new CB1 and I have followed the directions but I have a MacBook so the instructions are very vague when it comes to steps on Mac OS system.  I've tried to find the CB1 and it won't show up anywhere on network even after flashing OS to it with the listed changes to the files.  So now the M5P will only show a constant light and one on the CB1 but no flashing power light on M5P now. When flashing the CB1 I had the RPIBOOT and USBOTG in on position as stated in manual. I've had the M5P powered via USB which has been plugged into my laptop, a usb charging adapter and even straight into a surge protector. 

What am I missing here? Does anyone have clear steps for a Mac user?  I'm very new to Klipper and have only been using Marlin on previous printers. This was an updated board for my E3P so I can run Klipper and use remote services.  Anyone's help would be really appreciated.  Thanks!

TimiSmalls, 2023-11-15:
> When flashing the CB1 I had the RPIBOOT and USBOTG in on position as stated in manual.
But you dont need to position these 2 when using a CB1 or? I think in the manual its just for a CM4 or?

Limey343, 2023-11-15:
> or?

Yes you have to have both the USBOTG and RPIBOOT along with the boot switch on the usbc as well.  However this CB1 will only work with the SD card in the slot, take it out and nothing.  It is the eMMC version so shouldn't need the sd card, however I have yet to figure out how to write to the eMMC as instructions are very limited.  Plus I have a MacBook Pro but it runs MacOS and in the manual it says not supported for running the sunxi -fel, in the manta 8 manual it says to run the rpiboot command to access the CB1 eMMC but it fails to show up at all as a large capacity storage device so I'm unable to write the OS image to it via the raspberry PI imager or any other means.  The CB1 is version 1.2 which I've heard had issues and now they sr won version 2.*.  I may just have to send this one back and see if I can get the most up to date versions CB1 eMMC or just go to a regular CB1 but I wanted the onboard storage.