bigtreetech/Manta-M5P #20: MIN1 & MIN2 endstops not triggering with TMC5160

Birkemosen, 2023-10-27:
I'm unable to get MIN1 (PD3) and MIN2 (PD2) to trigger in Klipper, using a configuration of 5 steppers (tmc5160).
MIN3 and MIN4 works.

M5P is configured with USB-to-CAN (PA11/PA12) and CAN on (PD0/PD1)
TMC51650 is configured for physical endstop (no jumpers installed on diag pins)
There is 3.3v between Signal & Ground pins on MINx ports.

Endstopswitch for testing is a microswitch in NC configuration.

Is this a board related issue or a Klipper issue ?

samuelleb11, 2024-12-19:
I'm having the same issue with TMC2209's. For now I had to put my endstops on MIN3 and MIN4 but I have an SFS 2.0 to connect and I can't because MIN1 and MIN2 are not working for me. Did you solve the issue ?

Birkemosen, 2024-12-21:
> I'm having the same issue with TMC2209's. For now I had to put my endstops on MIN3 and MIN4 but I have an SFS 2.0 to connect and I can't because MIN1 and MIN2 are not working for me. Did you solve the issue ?

I never solved the issue, I switched to sensorless homing as a workaround.

liampwll, 2026-11-06:
BTT submitted a fix for this to Klipper: https://github.com/Klipper3d/klipper/pull/6462