bigtreetech/Manta-M5P #25: Wrong MCU Firmware configuration

jayxcfgj, 2024-03-19:
In the actually documentation V1.0 are the wrong MCU Firmware configuration. 
There are declared a bootloader offset with 8 KiB. 
This creates issues and after compile tthis you can't get the serial of the board. 

Please correct the doku to the correct working setting "no bootloader".

palouf34, 2024-05-20:
Do you have a solution to reflash the firmware because I made the mistake and now I'm trying to go through the sd card but the firmware.bin file is not taken into account when the card is completely rebooted.

jayxcfgj, 2024-05-20:
@palouf34 you can try to reflash the firmware through the DFU
like this:
https://www-drklipper-de.translate.goog/doku.php?id=klipper_faq:flash_guide:stm32h743:btt_manta_m8n_usb&_x_tr_sl=auto&_x_tr_tl=de&_x_tr_hl=de&_x_tr_pto=wapp

palouf34, 2024-05-20:
@jayxcfgj 

thank's for this solution.

aslesar2015, 2024-06-06:
For those who lost the original bootloader, here's a [copy of mine](https://github.com/aslesar2015/Btt_Manta-M5P_Bootloader). You can upload it to CB1 using WinSCP, MobaXterm or something similar (e.g. to the ~/klipper/out directory). After that:
1. Put the board into DFU mode.
2. Check with the `lsusb` command that the board has successfully entered DFU mode.
3. Warning, this will remove the current firmware!! Run the command `sudo dfu-util -a 0 -d 0483:df11 --dfuse-address 0x08000000:force:mass-erase -D ~/klipper/out/Manta_M5P_bootloader.bin`.