bigtreetech/Manta-M5P #35: USB-C port for connecting peripherals

WayneManion, 2025-01-22:
Hello. I tried to attach peripherals to my Manta M5P using the USB-C port on the M5P without success. I tried moving the "UART-to-USB"/"USB-OTG" switch, but I never measured any voltage on the 5V wire. 

Thanks

looxonline, 2025-01-22:
What kind of peripherals? The USBC port is only for serial data to and from the CB1/2 module or for programming the eMMC on the module. There will be no voltage on the wire unless you install the jumper but the jumper should only be installed when programming eMMC in order to avoid a clash between the integrated regulator and the USB voltage input (if any). It is not designed to be a voltage output.

WayneManion, 2025-01-22:
I installed the M5P in my Creality K1 printer (along with an EBB42 and a Cartographer connected over canbus). I was hoping to use the USB-C port to connect a USB hub that in turn was connected to the stock camera and the USB port on the front of the printer. I was hoping to use USB-C instead of the USB-A ports because USB-C generally provides a more positive connection than USB-A.

Here is a picture of my little internal USB hub: 
[image]

What I would REALLY like is if BTT started including a USB pin header on boards so I could connect USB devices using a 4-pin JST-XH connector. This applies to SoC-hosting boards like the Manta M5P as well as toolhead boards like the EBB42.

looxonline, 2025-01-22:
Some of the motherboards include a JST USB connector. I believe that the M8P is one of them but I am not sure about the M5P. You can check the pin diagram. The USB-C connector does not work like you want it to. The USB-A connector is perfectly suited to this application.

closed 2025-01-22 by looxonline

blackICE19240, 2026-01-01:
Was just caught by the same thing 😅