bigtreetech/Manta-M5P #37: PLEASE HELP ME

Pitsweer, 2025-02-21:
So I've got the manta M5P with CB1, installed klipper and all went great, until I needed to get something that looks like serial: /dev/serial/by-id/usb-Klipper_Klipper_firmware_12345-if00

I'm stuck at ls: cannot access '/dev/serial/by-id': No such file or directory

i've looked everywhere and nothing helps.

WayneManion, 2025-02-21:
I believe you need to build and flash Klipper to the MCU before it will show up that way in /dev/serial/by-id/

Pitsweer, 2025-02-22:
Yes...but how? I've tried everything.

RichBlac, 2025-02-24:
I had the same issue with my Manta M5P and a CM4. What version board do you have? Mine is version 1. I think these first versions were a crapshoot at best, and we unfortunately got the shit end of the deal. Very frustrating and dont expect BTT to do anything about it either. They will just give you the run around and piss you off more. Typical unaccountable people wont stand behind their shit products.

RichBlac, 2025-02-25:
I switched from a cm4 to a cb1 and now I can't get power to the cb1. There is only the red led on the mcu that is lit up now. None of the green led's on the the mcu or cb1 are on. I also have the m4p manta and have had zero issues with it. But this m5p is a total turd burger

Pitsweer, 2025-03-05:
Got this sorted...after I got CB2.