bigtreetech/Manta-M5P #39: Need help again please.

Pitsweer, 2025-03-05:
So I got everything sorted including setting up CanBus. 

Now new issue came up. Wired up heat bed using SSR. Temp shows room temp. When I do PID tuning or just set the temp to say 60°C all it does is goes up maybe 2°C and drops to room temp again until is shutdown due to heater error. I know the SSR works because I was using it on a different printer running Duet electronics. Thermistor set to Generic 3950 as it's a silicone keenovo heater. Connected to the Heat bed pin on M5P. (The one next to the DC in)

bradtaylor96, 2025-04-10:
Make sure you have 24volts going to the bed-in screw down terminals. The bed heater output is just a relay that closes the BED-IN and BED-OUT ports.

Pitsweer, 2025-04-11:
I got it sorted. I had the positive and negative the wrong way going to the SSR.

closed 2025-04-11 by Pitsweer