bigtreetech/Manta-M5P #40: DSI instructions in the manual not working

OriginaldCrypt, 2025-03-13:
I just built a Manta M5P + CM4 + 7 inch DSI screen and followed the instructions in the manual here:

https://github.com/bigtreetech/Manta-M5P/blob/master/BIGTREETECH%20MANTA%20M5P%20V1.0%20User%20Manual.pdf

> DSI1 Display
> The default display interface is HDMI, and the DSI interface of MANTA M5P is DSI1, you need to download the DSI1 driver, and enter the following in the command line: sudo wget https://datasheets.raspberrypi.com/cmio/dt-blob-disp1-cam1.bin -O /boot/dt-blob.bin After downloading this driver and restarting, the screen connected to the DSI interface can be displayed normally. If you want to use the HDMI interface, you need to delete the downloaded /boot/dt-blob.bin driver and restart, and then the HDMI can output normally.

but it does not work. The only way to make it work is by including the following lines in /boot/firmware/config.txt

dtoverlay=vc4-kms-dsi-7inch-overlay,7_0_inchC,i2c1

In a similar procedure as described for the WaveShare DSI 7" screen

BR

bigtreetech, 2025-03-13:
Thanks for the heads up. It seems that in the update from FKMS to KMS by RPi, the device tree overlay blob is no longer required. We will update docs.

bigtreetech, 2025-03-27:
Fixed in fb4dd2d300afd2e2ef7c326f6db2c6eaba3c1bf4. For the Raspberry Pi 2023-10-10 and later systems, the configuration path changes from '/boot/' to '/boot/firmware'.

closed 2025-03-27 by bigtreetech

OriginaldCrypt, 2025-03-27:
Are you sure that the rest of the configuration changes is still correct? I tried also by putting the dt-blob file inside /boot/firmware and it didn't work. The way I made it work was by including the following lines in /boot/firmware/config.txt

dtoverlay=vc4-kms-dsi-7inch-overlay,7_0_inchC,i2c1

It looks like for current distribution an overlay is already present, it is only required to be  activated.

BR

bigtreetech, 2025-03-27:
> Are you sure that the rest of the configuration changes is still correct? I tried also by putting the dt-blob file inside /boot/firmware and it didn't work. The way I made it work was by including [...]

I downloaded the latest system(`2024-11-19`) from the [official website](https://downloads.raspberrypi.com/raspios_lite_armhf/images/raspios_lite_armhf-2024-11-19/2024-11-19-raspios-bookworm-armhf-lite.img.xz), 

Write it into the CM4 eMMC version, and the directly power on the device, the DSI screen will not display any content, connect to ssh  terminal via a ethernet after the device starts, and then run:
```
sudo wget https://datasheets.raspberrypi.com/cmio/dt-blob-disp1-cam1.bin -O /boot/firmware/dt-blob.bin
```
[image]
After the download is complete, `sudo reboot` the device.
After restarting, the screen can display content normally
[image]

bigtreetech, 2025-03-27:
Can you share your system image link so that I can test with the same version? 

By the way, On my system Image, It should be
```
dtoverlay=vc4-kms-dsi-7inch,7_0_inchC,i2c1
```
not `vc4-kms-dsi-7inch-overlay`.

In this way, the DSI screen can also display normally on my side, but it seems that the screen starts to display after the Linux kernel is started, while the `/boot/firmware/dt-blob.bin` method seems to be that the DSI screen can display the rainbow screen + log information in the uboot stage after startup.

And thank you very much for your feedback.

OriginaldCrypt, 2025-03-27:
My image is the same, but "full" version

https://downloads.raspberrypi.com/raspios_full_arm64/images/raspios_full_arm64-2024-11-19/2024-11-19-raspios-bookworm-arm64-full.img.xz 

Regarding the overlay image name, it could be as you describe. I can't confirm as my CM4 was damaged and it won't boot anymore.

BR

thomasfjen, 2025-03-30:
Thanks for the updated instructions @bigtreetech. 
Would it be possible to add pictures of how the cable is to be inserted in the manual? 
I struggled with that for days, with your latest instructions and turning around the cable 180° it finally worked.
Currently you can only guess how that cable is inserted, the contacts of the dsi cable should face the manta pcb.

And can you add pictures of the different type of cables for the csi/dsi connector as the CSI is a TYPE A and DSI a TYPE B Cable?